alchitry/Alchitry-Labs-V2 #6: repeat loop usage shows different behaviour

natalieagus, 2024-07-09:
Hi there, I am trying to implement a simple ripple carry adder which requires a repeat loop to chain the full adder modules together.  Depending on how I set the indexes, they don't work quite the same even though they should logically work the same based on the [documentation](https://alchitry.com/tutorials/lucid-reference/#repeat) on your site. 

First of all, a plain connection between `fa` (full adder) units work as expected:

```
// rca.luc
// A ripple-carry-adder combinational logic unit
module rca #(
    SIZE = 2 : SIZE > 1
)(
    input a[SIZE],
    input b[SIZE],
    input cin,
    output cout,
    output s[SIZE],
) {

    fa fa[SIZE];
    
    always {
        fa.a = a;
        fa.b = b;
        fa.cin[0] = cin;
        
        // plain connection works 
         fa.cin[SIZE-1:1] = fa.cout[SIZE-2:0]
        
        cout = fa.cout[SIZE-1];
        s = fa.s;

    }
}
```

The `fa` unit is just a simple 1-bit full adder module:
```
// A full adder combinational logic unit 
module fa (
    output s,
    output cout,
    input a,
    input b,
    input cin
) {
    always {
        s = a ^ b ^ cin;
        cout = (a & b) + (a & cin) + (b & cin);
    }
}
```

I try to replace each `fa`'s `cin, cout` connection with a `repeat` loop, and came up with 3 versions that should (logically) works the same. 

```
        // version 1
         repeat(i, SIZE-1, 0){
            fa.cin[i+1] = fa.cout[i];
        }

        // version 2
        repeat(i, SIZE-1, SIZE-1, -1){
            fa.cin[i] = fa.cout[i-1];
        }

       // version 3
        repeat(i, SIZE-1, 1){
            fa.cin[i] = fa.cout[i-1];
        }
```

Here's the observed behavior:
- **version 1**:  works in simulation, for all values of `SIZE` > 2. This behaves the same as plain connection above. 
- **version 2**: if I set `SIZE` as 3, the following error shows. There's no error with **other** `SIZE` value e.g: `2,4,5,6...`. However, simulation shows the **wrong** output,  e.g: 1+1 does not give 2, but gives 10. 
[image]
- **version 3**:  same behavior as **version 2**

I restarted Alchitry lab (just in case there's simulation cache), when testing each version. 

I uploaded the test project [here](https://github.com/natalieagus/test-repeat-loop-lucid-v2) for your reference. Thank you.

natalieagus, 2024-07-09:
Update: **version 1** simulation output is not consistent.

If `SIZE` in `alchitryTop.luc` is set to 3, then the simulation output shows the correct addition between two 3 bit numbers. 

However, if `SIZE` is set to > 3, then the sum output shows the wrong value, e.g: 1+1 is 10 and not 2.

natalieagus, 2024-07-09:
Here's another bug with the `repeat` loop. Suppose I have a `mux2` module:

```
module mux2 (
    input s0,
    input in[2],
    output out
  ) {

  always {
    out = in[0];
    case (s0) {
      0: out = in[0];
      1: out = in[1];
    }
  }
}
```

I want to use this to create a N-bit left shifter of an 8-bit sample input. For testing purposes, my `alchitryTop` contains:

```
      mux2 shift_unit[8];
      sig in_1_shift_unit[8];
      const SHIFT = 2;
```

I use `ioDip[1][0]` to toggle between `shift` or `no shift`. I can enter the 8-bit input via `ioDip[0]`. 

```
        shift_unit.s0 = 8x{ioDip[1][0]};
        in_1_shift_unit = c{ioDip[0][8-1-SHIFT:0], SHIFTx{0}};
```

If I manually connect each mux as such, this works just fine in the Simulator:

```
        // works 
        shift_unit.in[0] = c{in_1_shift_unit[0], ioDip[0][0]};
        shift_unit.in[1] = c{in_1_shift_unit[1], ioDip[0][1]};
        shift_unit.in[2] = c{in_1_shift_unit[2], ioDip[0][2]};
        shift_unit.in[3] = c{in_1_shift_unit[3], ioDip[0][3]};
        shift_unit.in[4] = c{in_1_shift_unit[4], ioDip[0][4]};
        shift_unit.in[5] = c{in_1_shift_unit[5], ioDip[0][5]};
        shift_unit.in[6] = c{in_1_shift_unit[6], ioDip[0][6]};
        shift_unit.in[7] = c{in_1_shift_unit[7], ioDip[0][7]};
        
        ioLed[0] = shift_unit.out
```

[image]

However, the moment i use repeat, I have weird output. Even when the selector is set to be "not shifting", it couldn't pass the original signal from `ioDip[0]` as output. 

```
        // has bug
         repeat(i, 8, 0, 1){
            shift_unit.in[i] = c{in_1_shift_unit[i], ioDip[0][i]};
         }
        
        ioLed[0] = shift_unit.out
```

[image]

alchitry, 2024-07-10:
Thank you for this excellent write up! I was able track down the issue and fix it this morning. It actually didn't have to do with the repeat loop but rather the indexing inside the loop.

I don't think I've documented this yet, but signals inside expressions are resized to be the size of the final width before being evaluated. The final width value was incorrectly being set to 1 bit for bit selections making [i] only able to select [0] or [1].

commit 15393ff6 (2024-07-10): Fixes #6; Issue was with width fences not being set for bit selections.
closed 2024-07-10 by alchitry