alchitry/Alchitry-Labs-V2 #15: Unable to Build AlchitryLabsV2 Project With Vivado 2023.2 on Debian 12.5: Vivado can't launch

natalieagus, 2024-08-06:
Hi there, I was trying to build a basic project on Debian 12.5 with Vivado for Alchitry Au. I was always able to build one with Alchitry Labs 1.2.7 successfully (bottom left), but with AlchitryLabsV2 2.0.11 Preview it fails to launch Vivado:

[image]

I have set the Vivado path to be the same as I did in 1.2.7, which is `/mnt/vivado/Vivado/2023.2`. The bin exists as shown in the screenshot, and AlchitryLabs1.2.7 was able to find it. I ran AlchitryLabsV2 with root permission too but to no avail. 

Here's the versions used:
1. AlchitryLabsV2 2.0.11 Preview
2. Debian 12.5 
3. Java: OpenJDK17 
4. Vivado 2023.2 

Was wondering if there's incompatibility in the versions. Thanks!

natalieagus, 2024-08-06:
Update: If i compile from source, I manage to get it to run Vivado. The above was from https://github.com/alchitry/Alchitry-Labs-V2/releases/download/2.0.11/alchitry-labs-2.0.11-linux-amd64.tar.gz I tried other versions as well: 2.0.10. Same issue. Only if I compile the editor from scratch then it is able to run Vivado. 

However I run into another synthesis error. Will open another issue for that.

alchitry, 2024-08-12:
This seems to be an issue with the included JRE ([maybe permissions](https://stackoverflow.com/questions/61301818/java-failed-to-exec-spawn-helper-error-since-moving-to-java-14-on-linux). The pre-built code ships with a JRE while you're using your own JRE when you compile from source. 

I'm firing up a VM to test this out as I can't reproduce the issue on my desktop (Arch with Gnome).

alchitry, 2024-08-12:
I was unable to reproduce this on Ubuntu 22.04.

Check the permissions on the file `alchitry-labs-2.0.11/lib/runtime/lib/jspawnhelper` to see if it can be executed.

natalieagus, 2024-08-13:
Yes, jspawnhelper is executable. 

[image]

I specifically use this VM: https://mac.getutm.app/gallery/debian-12-rosetta so that I can compile on my Apple Silicon Mac, and then load the binary to my Alchitry Au using Alchitry Loader from my host OS. 

I can try with other native x86 VMs later on, using EC2 instance or something because I don't have the hardware with me. For now, compiling from source works just fine. Thank you.

natalieagus, 2024-08-15:
Update: it was OK with other VMs, seems like it's specific somehow to this Debian 12.5 + Rosetta VM build. Will update if I ever find out why.

alchitry, 2024-08-28:
I'm going to close this since it seems machine specific. If more people have this problem we can re-open it.

closed 2024-08-28 by alchitry