alchitry/Alchitry-Labs-V2 #19: binToDec bug: 7seg in Alchitry Au does not display the intended value

natalieagus, 2024-08-21:
Thank you for the details on the previous issue I opened. 

I have pulled the [latest commit](https://github.com/alchitry/Alchitry-Labs-V2/commit/08d1d6b340a8cf31a9976443dcfad2130582d600) you have and used it to compile a simple project utilising `binToDec` from Alchitry's component library, but unfortunately I ran to two issues:
- In hardware, the 7seg does not display the expected number properly when it receives `.digits` from `binToDec` module. None of the 7segs are lit up. 
- Simulation also does not work as expected using the default `binToDec` component library, but I did a little tweak (`V1` below) and it did show the right 7seg value only on simulation (but not hardweare) 

I have the project in my repo [here](https://github.com/natalieagus/binToDecTest.git). The Alchitry Au's 7 seg is already tested and confirmed to be not faulty. 

## Simulation Discrepancy 

The library version of `binToDec.luc` does not display the expected value on the 7seg while this V1 version that I adapted from previous Alchitry Lab V1 library **works** in Alchitry Labs 2.0.12 Simulation: 

```
module binToDecV1 #(
    DIGITS ~ 1 : DIGITS > 0 && DIGITS < 20,           // limited by 64 bit constants in the tools
    LEADING_ZEROS = 0 : LEADING_ZEROS == 0 || LEADING_ZEROS == 1
)(
    input value[$clog2($pow(10, DIGITS))],            // minimum number of bits for DIGITS 
    output digits[DIGITS][4]                          // decimal output
) {
    
    sig scale[$width(value)]
    sig remainder[$width(value)]                         // running remainder
    sig sub_value[$width(value)]                         // temporary subtraction value
    sig blank                                          // flag for leading zeros
    
    always {
        repeat(i, DIGITS){
            digits[i] = d11                                // default to invalid number
        }
        
        remainder = value                                // initialize remainder
        blank = !LEADING_ZEROS                           // set blank zero flag
        scale = 0
        sub_value = 0
        
        if (value < $pow(10, DIGITS)) {                   // if can be displayed
            repeat(j, DIGITS, DIGITS-1, -1){// for each digit  
                scale = $pow(10, j)                          // get the scale for the digit
                
                if (remainder < scale) {                      // if this digit is 0
                    if (j != 0 && blank)                        // use 10 for blank
                        digits[j] = 10
                    else                                        // or 0 for zero
                        digits[j] = 0
                } else {                                      // digit is 1-9
                    blank = 0                                  // don't blank future zeros
                    sub_value = 0                              // default to no subtraction
                    repeat(i, 9, 9, -1){// for each possible value (starting from 9)
                        
                        if (remainder < (i+1) * scale) {          // if remainder is less than value
                            digits[j] = i                          // set digit to this value
                            sub_value = i * scale                  // set subtraction value
                        }
                    }
                    remainder = remainder - sub_value          // subtract off last digit
                }
            }
        }
    }
}
```

Here's the current Alchitry Lab V2 `binToDec` component library for easy reference: 
```
module binToDec #(
    DIGITS ~ 1 : DIGITS > 0 && DIGITS < 20,           // limited by 64 bit constants in the tools
    LEADING_ZEROS = 0 : LEADING_ZEROS == 0 || LEADING_ZEROS == 1
)(
    input value[$clog2($pow(10, DIGITS))],            // minimum number of bits for DIGITS
    output digits[DIGITS][4]                          // decimal output
) {
    always {
        repeat(i, DIGITS)                                   // for all digits
            digits[i] = d11                                 // default to invalid number

        sig remainder[$width(value)] = value                // initialize remainder
        sig blank = !LEADING_ZEROS                          // set blank zero flag

        if (value < $pow(10, DIGITS)) {                     // if can be displayed
            repeat(j, DIGITS, DIGITS-1, -1) {               // for each digit
                sig scale[$width(value)] = $pow(10, j)      // get the scale for the digit

                if (remainder < scale) {                    // if this digit is 0
                    if (j != 0 && blank)                    // use 10 for blank
                        digits[j] = 10
                    else                                    // or 0 for zero
                        digits[j] = 0
                } else {                                    // digit is 1-9
                    blank = 0                               // don't blank future zeros
                    sig sub_value[$width(value)] = 0        // default to no subtraction
                    repeat(i, 9, 9, -1) {                   // for each possible value (starting from 9)
                        if (remainder < (i+1) * scale) {    // if remainder is less than value
                            digits[j] = i                   // set digit to this value
                            sub_value = i * scale           // set subtraction value
                        }
                    }
                    remainder = remainder - sub_value       // subtract off last digit
                }
            }
        }
    }
}
```

I fail to see any theoretical differences between the two. 

When i set the `DIV` rate of `multiSevenSeg` to be low enough for simulation:
```
multiSevenSeg seg(#DIV(3))
```

And declare each instance of the module in `alchitryTop`:

```
    binToDec binToDecOriginal(#DIGITS(4))
    binToDecV1 binToDecV1(#DIGITS(4))
```

Then set its value to some constant:

```
        binToDecOriginal.value = 4b0101
        binToDecV1.value = 4b0111
```

I observe the following in Simulation: 

_Note: I use `ioDip[0][0]` and` ioDip[0][1]` to toggle display of output from binToDecOriginal or binToDecV1_

```
        if (ioDip[0][0]){
            seg.values = binToDecOriginal.digits
            ioLed[0] = binToDecOriginal.digits[0]
            ioLed[1] = binToDecOriginal.digits[1]
            ioLed[2] = binToDecOriginal.digits[2]
        }
        
        if (ioDip[0][1]){
            seg.values = binToDecV1.digits
            ioLed[0] = binToDecV1.digits[0]
            ioLed[1] = binToDecV1.digits[1]
            ioLed[2] = binToDecV1.digits[2]
        }
```
**When ioDip[0][0] is 1**: library component `binToDecOriginal` does not lead to the right constant in simulation's 7seg, which is supposed to be 5. On the contrary, none of the segments is on:

[image]

**When ioDip[0][1] is 1**: Simulation shows correct value on 7seg using `binToDecV1`, which is set to constant 7 for illustration:
[image]

`ioLed[2:0]` display digit 2, 1, and 0 respectively (from `binToDecOriginal/V1.digits`). 

## When Compiled and Run on Alchitry AU 

Both `binToDecOriginal/V1` does not show the correct output on the 7seg (off). Upon inspection, all values of `binToDecOriginal/V1.digits` is `4b1011`.

alchitry, 2024-08-21:
Thanks again for the excellent write up. It's super helpful being able to just clone the project and reproduce it immediately.

closed 2024-08-21 by alchitry

alchitry, 2024-08-21:
This was caused by two bugs.

The first was during simulations the local signal's values weren't being updated if the value came from the line it was declared on. For example, `sig remainder[$width(value)] = value` would only be evaluated on the first pass so as `value` changed, `remainder` didn't. This was fixed with https://github.com/alchitry/Alchitry-Labs-V2/commit/2669aa441b25a7428111303354f1530fcdf1da4c

The second bug was caused by bad Verilog translation for repeat blocks with negative steps. I'm actually a little confused why it built at all but this was fixed in https://github.com/alchitry/Alchitry-Labs-V2/commit/eed692ea2b1214c92ee62f5dbe167969f8ede675

Repeat blocks now always translate to the basic `i = 0; i < steps; i = i + 1` format and the step and start values are calculated and written to another signal independently. I need to test the behavior with negative values more still.

alchitry, 2024-08-21:
Unrelated to the bug, but I added a `$isSim()` function that evaluates to `1b1` when simulating and `1b0` all other times. This is useful for stuff like 
```
multiSevenSeg seg(#DIV($isSim() ? 3 : 16)) // 3 for simulation, 16 for hardware
```

natalieagus, 2024-08-22:
Thank you very much for the information and fixes. 

The `$isSim()` function is very useful too, thanks!

commit 4cd19bfc (2024-08-21): Fixes #19 - Fixed issue with for loops using a negative step size. Fixed issue with signed constants not converting to Verilog correctly.
commit eed692ea (2024-08-21): Fixes #19 - Fixed issue with for loops using a negative step size. Fixed issue with signed constants not converting to Verilog correctly.