alchitry/Alchitry-Labs-V2 #21: Failed to convert source file to Verilog: signal "binToDec.digits" was expected to be driven by this always block but it wasn't.

natalieagus, 2024-09-05:
I pulled and build your latest commit: v2.0.13 [34e21f28143294ec147b6cb81116e0c49c3a08e2](34e21f28143294ec147b6cb81116e0c49c3a08e2) to compiled test repo I enquired couple weeks ago: https://github.com/natalieagus/binToDecTest

Simulation works completely but I encountered this error when trying to compile the code:

[image]

I was simply importing the given Binary to Decimal component library: `binToDec.luc`. I used it in the `AlchitryTop.luc` as follows:

```
   binToDec binToDecOriginal(#DIGITS(4))
```

Was wondering if there's any syntax change? Thank you!

natalieagus, 2024-09-05:
Additional info:

I created a fresh project using Blinker Demo as a template for Alchitry Au. Compiled it once as-is (successfully). 

Then, I simply imported the binToDec.luc component without using it in `alchitryTop.luc`. When I tried to compile, same error encountered:

[image]

Test repo: https://github.com/natalieagus/test-bintodec-plain.git

alchitry, 2024-09-06:
Thanks for this! It exposed a couple of bugs.

The repeat block should have satisfied the driver check since it always iterates over every value of `digits` but during building, the constant `DIGITS` is unknown.

```
        repeat(i, DIGITS){
            digits[i] = d11                                // default to invalid number
        }
```

This makes it really difficult to check if every element of `digits` is assigned. However, this doesn't actually need to be checked since the project is checked with all actual parameter values from the top down.

To resolve this, I added a skip to the driver check during the building stage.

I also fixed a bug when using the duplication syntax with parameter sizes. The repeat loop was replaced with the following but both work.
```
digits = DIGITSx{{4d11}}
```

commit c83af4b4 (2024-09-06): Fixes #21 - Fixed issues revolving around UndefinedWidths. Skip driver check when building.
closed 2024-09-06 by alchitry