alchitry/Alchitry-Labs-V2 #30: Unable to initialise arrays of module requiring arguments properly

natalieagus, 2024-09-25:
Tested on: Alchitry Labs V2 2.0.16 
OS: Debian 12.5 (sim & compile), macOS Sonoma 14.2.1 (sim) 
Repo of test code: https://github.com/natalieagus/buttonConditionerTest.git

In this test code, I connected all ioButton to buttonConditioner, and display `buttonConditioner.out` to the ioLed. The corresponding ioLed should **light up** when the ioButton is **pressed**. 

If i instantiate them one by one as follows, they seem to exhibit the expected behavior on both simulation and hardware. That is, the corresponding ioLed such as `ioLed[0][i]` will light up when `ioButton[i]` is pressed for `i = 0` to `4`. 

```
        // these works with hardware and sim
        buttonConditioner buttonCond0(#CLK_FREQ($isSim() ? 1000 : 10000000))
        buttonConditioner buttonCond1(#CLK_FREQ($isSim() ? 1000 : 10000000))
        buttonConditioner buttonCond2(#CLK_FREQ($isSim() ? 1000 : 10000000))
        buttonConditioner buttonCond3(#CLK_FREQ($isSim() ? 1000 : 10000000))
        buttonConditioner buttonCond4(#CLK_FREQ($isSim() ? 1000 : 10000000))
 

... 
       // always block 
        buttonCond0.in = ioButton[0]
        buttonCond1.in = ioButton[1]
        buttonCond2.in = ioButton[2]
        buttonCond3.in = ioButton[3]
        buttonCond4.in = ioButton[4]

       // output to ioLed[0]
        ioLed[0] = c{3b0, buttonCond4.out,buttonCond3.out, buttonCond2.out, buttonCond1.out, buttonCond0.out}

```

However when I instantiate them this way: 

```
     const CLK_FREQ = $isSim() ? 1000 : 10000000

    .clk(clk){
     buttonConditioner ioButtonCondTest1[5](#CLK_FREQ(5x{{CLK_FREQ}}))
```

or this way:

```
     buttonConditioner ioButtonCondTest2[5](#CLK_FREQ({CLK_FREQ,CLK_FREQ,CLK_FREQ,CLK_FREQ,CLK_FREQ})) 
```

and utilise + display its output as follows:

```
        ioButtonCondTest1.in = ioButton
        ioButtonCondTest2.in = ioButton

        // display
        ioLed[1] = c{3b0, ioButtonCondTest1.out}
        ioLed[2] = c{3b0, ioButtonCondTest2.out}
```

it won't work on hardware. By right, as shown in simulation, `ioLed[j][i]` should all light up when `ioButton[i]` is pressed for j=0,1,2 and i=0 to 4, shown in screen recording below: 

https://github.com/user-attachments/assets/0b1b80fb-d39f-4a5c-b159-2baaef1bc9c3

However in hardware, only ioLed[0][i] works. The other: ioLed[1] and ioLed[2] is always 0, which means that ioButtonCondTest1 and ioButtonCondTest2 both always output 0 for some reason. Here's a photo with ioButton[0] being pressed:

[image]

Was I initializing the array of buttonConditioner properly? Thanks.

alchitry, 2024-09-26:
There were some bugs introduced in modules arrays in the major build overhaul in 2.0.13. The commit I'm working on now is another major translator change that fixes this along with some other Verilog issues.

alchitry, 2024-09-26:
Note that there are some issues with the new system that still need to be ironed out. I'll be working on it tomorrow.

commit 7fa8cf06 (2024-09-26): Fixes #30. Implements arrays and structs natively in SystemVerilog.
closed 2024-09-26 by alchitry