alchitry/Alchitry-Labs-V2 #31: tclkit Failure

MkJAS, 2024-09-27:
In trying to use the iCEcube2 tool to build a project I got an error along the lines of "cant find a usable init.tcl" with a bunch of directory errors pointing to WindowsApps\Alchitry.AlchitryLabs_2.0.18.0_x64__wrkdgnpcgqe54\app

I was unable to build and was only able to solve the issue after replacing the tclkit-8.5.17.exe in WindowsApps\Alchitry.AlchitryLabs_2.0.18.0_x64__wrkdgnpcgqe54\app\bin with another version of the tclkit downloaded from the tclkit website.

alchitry, 2024-09-30:
Thanks for this. I've reproduced it and it seems that it isn't an issue with the file we have but rather the build tools messing with the file when it is being packaged. I've contacted Hydraulic (the guys who make the build tools) to see if they can help figure out what's going on.

alchitry, 2024-09-30:
It turns out the tclkit version that I was using would break when the .exe was signed. I got a version that survives code signing. This should be fixed in the next release.

closed 2024-09-30 by alchitry