alchitry/Alchitry-Labs-V2 #33: "Failed to convert source file to Verilog"

natalieagus, 2024-10-06:
I apologize in advance for dumping this bug as-is and giving you the entire project's source code to test like this because I am not sure what was causing the problem in the first place. I have this other project that emulate a toy-CPU (RISC variant): https://github.com/natalieagus/betaAlchitryV2

I am able to compile and run it on Alchitry Au with v2.0.18, but v2.0.19 gives me this error straight up. I am not sure how to proceed with this: 

[image]

Note: The simulation wouldn't run on any version either, that's fine because it still ran on the hardware for v2.0.18 and earlier. Could it be because I am using the simple_ram module for this project? The io_led is showing all red color:

[image]

This is what it's supposed to look like upon loading in hardware: 

[image]

alchitry, 2024-10-07:
This was caused by line 36 in `beta_manual_tester.luc`

Creating an array out of a parameter should be an error since the width of the parameter is flexible. This used to not crash if you had a test/default value for the parameter as it would use that width but this would silently fail if the actual value was larger than the test/default value.

You can change the `3x{{CLK_FREQ}}` expression to have an explicit width for `CLK_FREQ` with `3x{{$resize(CLK_FREQ,32)}}`. You can change 32 to be something as a reasonable max width.

It might be possible to lift this restriction in the future but it would require significant work. The current fix just reports the error correctly.

alchitry, 2024-10-07:
I should also mention that red LEDs in the simulator indicate a value of `x`.

You're correct in that this is likely from the simple_ram module. Verilog is currently ignored by the simulator. I'm planning to add proxies to Lucid. I'll open an issue for this.

closed 2024-10-07 by alchitry

natalieagus, 2024-10-08:
Oh I see, thank you. Apologies, I forgot that parameter can vary in width hence we cannot create an array out of parameters.

commit e93f5dae (2024-10-07): Fixes #33. Correctly catch errors with unresolvable array widths.