alchitry/Alchitry-Labs-V2 #36: Lucid V2: bin_to_dec.luc fails to build in Vivado 2024.1

dheijl, 2024-11-17:
This is where it fails:

```
INFO: [Synth 8-6157] synthesizing module 'bin_to_dec' [/home/danny/alchitry/IO_Experiments_2/build/vivado/IO_Experiments_2.srcs/sources_1/imports/source/bin_to_dec.sv:7]	
Parameter DIGITS bound to: 3'b100 	
Parameter LEADING_ZEROS bound to: 1'b0 
ERROR: [Synth 8-11587] invalid assignment pattern [/home/danny/alchitry/IO_Experiments_2/build/vivado/IO_Experiments_2.srcs/sources_1/imports/source/bin_to_dec.sv:27]
ERROR: [Synth 8-6156] failed synthesizing module 'bin_to_dec' [/home/danny/alchitry/IO_Experiments_2/build/vivado/IO_Experiments_2.srcs/sources_1/imports/source/bin_to_dec.sv:7]
ERROR: [Synth 8-6156] failed synthesizing module 'alchitry_top' [/home/danny/alchitry/IO_Experiments_2/build/vivado/IO_Experiments_2.srcs/sources_1/imports/source/alchitry_top.sv:7]
```
You can find a minimal project [here](https://github.com/dheijl/AlchitryAuTests/tree/main/IO_Experiments_2).

dheijl, 2024-11-18:
Apparently it's the use of the parameter DIGITS in line 42 in bin_to_dec.luc : `DIGITSx{{4d11}}` that produces Verilog that Vivado doesn't like.

alchitry, 2024-11-20:
Thanks for your bug report and easy to reproduce example!

closed 2024-11-20 by alchitry

dheijl, 2024-11-20:
You're welcome!

commit d5664fda (2024-11-20): Fixes #36.