alchitry/Alchitry-Labs-V2 #38: Problem building project, but similation works OK

dheijl, 2024-11-22:
[Link to project to reproduce](https://github.com/dheijl/AlchitryAuTests/tree/main/IO_Experiments_2)

`Failed to convert module "mul10_addbcd" to Verilog! Failed to convert source files to Verilog. 

This should be considered a bug!

Undefined width during verilog conversion!java.lang.IllegalStateException: Undefined width during verilog conversion!	
at com.alchitry.labs2.parsers.hdl.lucid.SystemVerilogConverter.verilogArrayWidths(SystemVerilogConverter.kt:99)	
at com.alchitry.labs2.parsers.hdl.lucid.SystemVerilogConverter.verilogArrayWidths(SystemVerilogConverter.kt:94)	
at com.alchitry.labs2.parsers.hdl.lucid.SystemVerilogConverter.addDffs(SystemVerilogConverter.kt:232)	
at com.alchitry.labs2.parsers.hdl.lucid.SystemVerilogConverter.exitModule(SystemVerilogConverter.kt:618)	
at com.alchitry.labs2.parsers.grammar.LucidParser$ModuleContext.exitRule(LucidParser.kt:701)	
at com.alchitry.labs2.parsers.ParseTreeMultiWalker.exitRule(ParseTreeMultiWalker.kt:111)	
at com.alchitry.labs2.parsers.ParseTreeMultiWalker.walk(ParseTreeMultiWalker.kt:46)	
at com.alchitry.labs2.parsers.ParseTreeMultiWalker.walk$default(ParseTreeMultiWalker.kt:10)	
at com.alchitry.labs2.parsers.hdl.lucid.context.LucidBlockContext.walk(LucidBlockContext.kt:214)	
at com.alchitry.labs2.parsers.hdl.lucid.context.LucidBlockContext.walk$default(LucidBlockContext.kt:213)	
at com.alchitry.labs2.parsers.hdl.lucid.context.LucidBlockContext.convertToVerilog(LucidBlockContext.kt:242)	
at com.alchitry.labs2.parsers.hdl.types.Module.convertToVerilog(Module.kt:29)	
at com.alchitry.labs2.parsers.ProjectContext.convertToVerilog(ProjectContext.kt:75)	
at com.alchitry.labs2.project.Project$build$2.invokeSuspend(Project.kt:316)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)	
at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:101)	
at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:589)			
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:832)				
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:720)		
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:707)`

alchitry, 2024-11-26:
If you're able to make the IDE crash then it's an issue and you should leave it open so I don't forget about it ;)

closed 2024-11-26 by alchitry

dheijl, 2024-11-26:
I stand corrected...

commit 08039a1d (2024-11-26): Fixes #38