alchitry/Alchitry-Labs-V2 #46: Why does Alchitry Au Template Io Demo V1 with Pulldown need fake pulldowns?

natalieagus, 2025-01-17:
**Version**: 2.0.12 BETA 

First of all, this is more like a question and not an issue (not sure). 

I created a new project using Alchitry Au + Io Demo V1 with pulldown template: 

[image]

However this created fake pulldown units:

[image]

My understanding is that only the Cu boards need the fake pulldowns but not the Au boards. Can I clarify? 

Thank you.

natalieagus, 2025-01-17:
If I create a project without the pulldown:

[image]

This creates two folders Fake Pulldowns and Pulldowns:

[image]

But the project constraint is still the v1 without the pulldown:

[image]

alchitry, 2025-01-17:
Thanks for sharing this issue! There are two bugs you caught.

First, I messed up when migrating the Io templates to use the new ACF format. I accidentally based the pull down version off the fake pull down version instead of the non-fake version.

Using the non-fake version and replacing the constraint file with the pulldown version is exactly what the pull down template should be.

The second bug was with the template copy code. It was using a pattern match of `startsWith()` for finding the files to copy so it was copying all the projects that had names starting with "Io Demo V1" which included "Io Demo V1 Fake Pull-Downs" and "Io Demo V1 Pull-downs". You can safely delete these folders as they're actually two more complete projects.

commit 14f0f5f3 (2025-01-17): Fixes #46
closed 2025-01-17 by alchitry