alchitry/Alchitry-Labs-V2 #49: Question: Build does not meet timing in design using 2.0.23, but it worked in previous version (or does it?)

natalieagus, 2025-01-19:
Thank you for closing the previous issues promptly, they are very helpful.

I have another small question to verify regarding the timing report (this is more of a question than an issue). 

I was compiling my ALU project that I used to test the earlier versions as well. [This is the repo](https://github.com/natalieagus/aluAlchitryV2.git), it is a private repo because I have to teach my students soon and I don't want them to obtain the answer freely, so I have invited you as a collaborator instead. 

**Background:**
I compiled this same project with the previous release (2.0.22) and it worked fine, tested on both hardware and simulator.  I also tested it with 2.0.19 and it worked perfectly (both cases).

I compiled it with the latest release 2.0.23 and it complained that project does not meet timing specification `Project built but failed to meet timing.`, the file with the problem is `multiplier.luc`. 

My question is: _has this unit always failed to meet timing specification all along or was it something else?_ 

The project was migrated from Lucid V1 (Alchitry Labs 1.2.7) and was extensively tested with no issue. However at the time, it was just a 16-bit multiplier, so I suspect that the 32-bit multiplier has failed to meet timing all along. 

From the commit 18ac029d8244a588a5df03ca8f6c94eb384e1d71 it seems that you are simply reporting it right? And the timing computation is done by Vivado. 

I tested it with 2.0.22 but perhaps I did not carefully see the logs enough and missed `Optimizations did not improve timing on the net.` logs. I will unroll and try it in the day.

alchitry, 2025-01-20:
It's likely that it was always marginal. Sometimes when a design is super close to working the tools will occasionally work but the smallest seemingly unrelated change will break it. 

2.0.22 started reporting if timing failed. You're correct that Vivado does the actual analysis and the addition was to just make it more obvious when it fails.

2.0.23 has a decent refactor of the Verilog translator that mainly reordered the way things were in the Verilog file to match the original Lucid. This fixed a few bugs but reordering could make a very marginal design fail.

I'll give it a go tomorrow with your code and see if it works with 2.0.22 but not 2.0.23 for me.

natalieagus, 2025-01-20:
Thank you. 

My bad, I tried again and remembered the version wrongly. **It worked with 2.0.21, not 22.** 

[image]

Upon inspecting the Vivado logs further, it did say that optimizations did not improve timing on the net and did not pass the timing regardless, so likely that the 32-bit multiplier design did not pass the timing checks in the first place. 

Thank you for your help. I will close this issue now.

closed 2025-01-20 by natalieagus