alchitry/Alchitry-Labs-V2 #51: No boards detected

djrscally, 2025-01-24:
Thanks for fixing the last issue, much appreciated. Next problem; when I connect my CuV2 Alchitry Labs and the loader both report no boards detected. The device powers on fine (and seems to be running some pre-configured blinky demo that cycles the LEDs sequentially). I am running Kubuntu 22.04. As best I can ascertain I need to make sure I have libusb installed, which is done with libusb-1.0-0. Running udevadm monitor and plugging the board in reports no issues. I am using an external USB hub, but tried a direct port to make sure that wasn't the cause and can confirm it's not.

Any ideas on how to debug would be very welcome.

alchitry, 2025-01-24:
It should just work with Ubuntu (or I would assume any of its derivatives).

Let's run through some troubleshooting steps.

1. Make sure you have a USB cable with data lines. Some USB cables only have power wires.
Run `lsusb | grep "0403"` This should print out something similar to the following.
```
Bus 001 Device 021: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
```
If you see this, then the board is connected.

2. Did you install the .deb or are you running from the files directly? You may need to setup a udev rule is you didn't use the .deb.
Check that the file `/ect/udev/rules.d/99-alchitry.rules` exists.
You can run `cat /etc/udev/rules.d/99-alchitry.rules` to print its contents. They should be the following.
```
SUBSYSTEM!="usb|usb_device", GOTO="ft2232_rules_end"
ACTION!="add", GOTO="ft2232_rules_end"

ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", RUN+="/bin/sh -c 'echo $kernel > /sys/bus/usb/drivers/ftdi_sio/unbind'", MODE="0666", GROUP="dialout", ENV{ID_MM_DEVICE_IGNORE}="1"

LABEL="ft2232_rules_end"
```

3. If things still aren't working, you could try running alchitry labs as root to rule out permissions issues. 
`sudo alchitry-labs`
If this works, there is probably an issue with the udev rule.

djrscally, 2025-01-27:
> Make sure you have a USB cable with data lines. Some USB cables only have power wires.

Baaaaaaaah embarrassingly this was the problem. Sorry to waste your time, I really should have checked that :/

closed 2025-01-27 by djrscally