alchitry/Alchitry-Labs-V2 #54: 2.0.26 beta upgrade: vivado ip broken (project not found)

dheijl, 2025-02-05:
Apparently something changed with the 2.0.26 update (PATH setting ?).

Projects build OK, but when I launch Vivado IP catalog, Vivado starts, but it doesn't show the project (sources window is empty).
I can then manually open the corresponding Vivado project and add some IP, but upon closing Vivado the new IP core is not picked up and no wrapper is generated.

I did not change Vivado versions, I'm still on 2024.1.

alchitry, 2025-02-05:
Does this happen with a project from a previous version of Alchitry Labs or a new one created with 2.0.26?

alchitry, 2025-02-05:
Did you have any cores generated from 2.0.21 or was 2.0.26 the first time you tried to add one to that project?

dheijl, 2025-02-05:
The project once had a clock wizard working but I removed it later, before upgrading to 2.0.26.

alchitry, 2025-02-05:
Ok I'll try and reproduce it. I'm installing Vivado in a VM now.

I changed some stuff in 2.0.25 related to Vivado paths to try and use relative paths where possible. I thought it would fix an error Vivado was giving me but it turns out that Vivado just has a bug in a Linux script that fails if there's spaces in path names (and it converts relative to absolute so nothing you can do there).

Deleting the cores isn't an issue (if they're not in the .alp file). I just wanted to make sure it wasn't some weird crossover issue from an old version to the new one.

dheijl, 2025-02-05:
The path on my Debian is `/home/danny/alchitry/teensy_spi`

alchitry, 2025-02-05:
I'm not able to reproduce this. I cloned your project then opened up the IP catalog and created a clk_wiz core.

```
No IP core project detected. Creating one...
/opt/Xilinx/Vivado/2024.2/bin/vivado -nojournal -nolog -mode batch -source /tmp/AlchitryAuTests/teensy_spi/cores/project.tcl
...
Launching Vivado..
/opt/Xilinx/Vivado/2024.2/bin/vivado -nojournal -nolog /tmp/AlchitryAuTests/teensy_spi/cores/managed_ip_project/managed_ip_project.xpr
...
Vivado exited...
Looking for cores...
Looking in clk_wiz_0...
  Found core .xci file!
  Found stub file!
Done.
```

Your settings look the same as mine. There isn't a top module since this is a detached IP only project.

dheijl, 2025-02-05:
You're right, Vivado does not show the project but I can add an IP and Labs V2 generates the wrapper.

Terribly sorry for wasting your time on this. 

I had previously added IP directly in Vivado and got confused when it did not show the project when launched from Labs...

closed 2025-02-05 by dheijl