alchitry/Alchitry-Labs-V2 #79: Unable to run Vivado on Alchitry

ndrew222, 2025-04-02:
If I try to hit Build on a test project for example, the output I get is
```
Starting Vivado...
application-specific initialization failed: couldn't load file "libxv_commontasks.so": libtinfo.so.5: cannot open shared object file: No such file or directory
```

If I try and do Vivado IP Catalog or Generate MIG Core (DDR), both give this output:
```
No IP core project detected. Creating one...
/home/andrew-fed/Applications/Xilinx/Vivado/2024.2/bin/vivado -nojournal -nolog -mode batch -source /home/andrew-fed/Documents/School/ehhh/cores/project.tcl
application-specific initialization failed: couldn't load file "libxv_commontasks.so": libtinfo.so.5: cannot open shared object file: No such file or directory
```

I'm not sure if this is an issue with Vivado or with Alchitry, but my install of Vivado 2024.2 is a fresh one.

ndrew222, 2025-04-02:
hmm fixed it by doing
```
sudo dnf install ncurses-compat-libs
```

alchitry, 2025-04-08:
This is a Vivado issue and nothing we can do about it. What OS were you using? If it wasn't on [Xilinx's official list](https://docs.amd.com/r/en-US/ug973-vivado-release-notes-install-license/Supported-Operating-Systems) you're on your own for getting libraries and stuff working.