alchitry/Alchitry-Labs-V2 #83: Alchitey Cu V2 not detected on Windows ARM64

tejasraman, 2025-04-23:
We recently bought an Alchitry Cu with an Io and have had some issues with getting it to work with Windows on ARM64.

The appropriate FTDI (VCP/D2xx) drivers are installed for the device and connecting to the serial terminal via PuTTY seems to function as intended.

However, Labs is unable to detect the device when it is plugged in on an ARM64 Windows laptop. It works on an x86_64 Windows laptop, as well as on x86_64 Linux.

The FTDIListDevices utility seems to present an error about the FTD2XX.dll file not being present on the ARM64 device, which is probably due to the fact that the DLL is for x86_64.

Is this DLL used within Labs and is there a chance the app could work on ARM64?

**Tested Hardware:**

- ARM laptop (@mandolinraman 's laptop): Surface Laptop 7 15", 64GB RAM, 1TB SSD, Windows 11 Home
- x86 laptop (my laptop): HP EliteBook Dragonfly G4, i7, 32GB RAM, 1TB SSD, Windows 11 Pro 64

alchitry, 2025-04-28:
As you've noticed, Windows ARM isn't currently supported. I'm happy to work on support but I don't have any hardware to test on. The issue you're seeing is because the D2xx implementation we rely on doesn't have ARM support.
https://github.com/alchitry/yad2xx/tree/d2xx_only_with_lib

It would need to be compiled for ARM then the .dll can be included. There's instructions here https://github.com/aushacker/yad2xx/wiki/Building-on-Windows and it might be possible for me to crosscompile.

tejasraman, 2025-04-28:
> As you've noticed, Windows ARM isn't currently supported. I'm happy to work on support but I don't have any hardware to test on. The issue you're seeing is because the D2xx implementation we rely on [...]

There is an Arm64 GitHub Actions runner now (see https://docs.github.com/en/actions/using-github-hosted-runners/using-github-hosted-runners/about-github-hosted-runners) so it may be possible to compile natively that way as opposed to cross compiling. I would be happy to try and compile it myself.