alchitry/Alchitry-Labs-V2 #84: Can't open on MacOS 13.4.1

tpolecat, 2025-04-26:
Hi, what are the system requirements for running on a Mac? The installer fails with `kLSIncompatibleSystemVersionErr` on MacOS Ventura 13.4.1 (see below). Thanks!

```
$ curl -L https://github.com/alchitry/Alchitry-Labs-V2/releases/latest/download/launch.mac | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  1492  100  1492    0     0   2948      0 --:--:-- --:--:-- --:--:--  2948
Downloading Alchitry Labs ...
######################################################################## 100.0%
Extracting to /Applications ...
The application cannot be opened for an unexpected reason, error=Error Domain=NSOSStatusErrorDomain Code=-10825 "kLSIncompatibleSystemVersionErr: The app cannot run on the current OS version" UserInfo={_LSLine=4101, _LSFunction=_LSOpenStuffCallLocal}
$ 
```

alchitry, 2025-04-28:
The minimum version is set at `13.5.0`. I'm not 100% sure where that came from to be honest so if it's a problem let me know and I'll look into seeing if it can be lowered.

closed 2025-07-11 by alchitry

tpolecat, 2025-07-11:
It worked! Thank you :-)