alchitry/Alchitry-Labs-V2 #96: scrolling lucid file get " index (33) out of range [0:4]" error

petera42, 2025-08-12:
revisiting stable Lucid source after release  Alchitry 2.0.40  
scrolling any lucid file results in index out of range error

last good look at sources  was probably with 2.0.37 
vivaldi invoked by 2.0.40 works fine

petera42, 2025-08-12:
Setting focus to any line and scrolling several lines down causes error
Using Win 10 .  Since this is not generally seen, I will treat it as a local issue and poke around.
Will report back.

dheijl, 2025-08-12:
Debian 12, same problem: click on a line anywhere, cursor up, cursor up, and immediately the following pop-up appears:

[image]

and Alchitry vanishes.

alchitry, 2025-08-12:
Thanks for the reproduction tips. I'll look into it this morning and get a new version out.